fossasia/susi_linux #5: Add Hotword Based Wakeup

betterclever, 2017-05-15:
Now, that prototype is almost ready, next step is to add Hotword based wakeup.

mariobehling, 2017-05-20:
Please follow up here, what technologies you are using and experimenting with.

betterclever, 2017-05-20:
**Already tested**
CMU Sphinx (Pocketsphinx) -
Issues - 
1. Slow recognition.
2. Not easy to train via Audio Samples (Like we train Google Assistant: by speaking "Ok Google" 3 times)
3. False positives on slow dictionary.
4. Negative results/ wrong detection on large dictionary.

Currently Testing:
Python Hotword Recogniton (https://github.com/sakethgsharma/HotWordDetection)
Advantages:
1. Training based recognition
2. Fairly correct results if training done right.

Problems being faced:
1. False positive results sometimes.
2. Figuring out a way for continuous detection.
3. Error while detection after training sometimes. (Will update with logs)

betterclever, 2017-05-28:
Pertaining to problems in Python Hotword Recognition Library, I switched back to Pocketsphinx for hotword. 
After, some adjustments in kws threshold value, the detection is fine now (mostly).
Yet, to eradicate false positive results though.

PR #10 (2017-06-06): Enabled Hotword Detection
Fixes #5 
**Changes**
- Added Hotword Detection using Pocketsphinx for handsfree interaction.
- Refactored folder and file names to be more meaningful.
closed 2017-06-06 by mariobehling