fossasia/susi_linux #9: Best Operating System suggestion for Hardware Application

betterclever, 2017-05-29:
As per current prototype of Susi Hardware that has hotword detection and handsfree interaction, any Operating System can be used for basic testing.
But for final product, we must decide and focus our work on a particular flavor of Linux, since it will help to maintain uniformity in project.

## Current Options
### Raspbian (Fork of Debian)
**Pros**
- Excellent support for Raspberry Pi development board.
- Most developers are familiar with Debian.

**Cons**
- Focused on Raspberry Pi, ignores other development boards.
- A lot of unneeded software preinstalled which uses RAM in the background.  

### Ubuntu Core
**Pros**
- Support for many development boards.
- Familiarity with Debain

**Cons**
- Packages for voice tts Festival which is being used is needed to be compiled manually,
- Voice packs for TTS are needed to be installed manually,

### Arch Linux for ARM/ Arch Linux
**Pros**
- Minimal. No extra software by default. Installation can be customized to suit our Susi Hardware Apps specific needs.
- Festival TTS Application is available on official repositories for all platforms (ARM/x86/x64) with voice packages as well. (HUGE)
- Fully Packaged solution can be distributed using AUR (https://aur.archlinux.org/) for installation as easy as a single line. (HUGE)

**Cons**
- Unfamiliarity of some developers with Arch Environment.
- Some different conventions from other operating systems like Python3 as default and bleeding edge distribution (always latest version of all packages). 

This being said, I strongly recommend Arch Linux to be used to develop solution however I'm free to other opinions as well.

betterclever, 2017-05-29:
@Orbiter @mariobehling Please guide.

mariobehling, 2017-06-05:
We don't need to decide this now. Postpone the final decision.

Orbiter, 2017-06-06:
From my point of view, a debian release should be available for all different plattforms, RPi, OrangePi, etc. While for The RPi that is Raspian the release may be different for other plattforms but I believe a debian should be there always.

betterclever, 2017-06-06:
Okay. I found **armbian** : https://www.armbian.com/ . It supports majority of boards.
Also for Raspbian for Raspberry Pi, there is a lite version without unneeded software as mentioned above, 

I am currently working on Arch Linux for ARM, I'll shift work to Debian.

betterclever, 2017-06-10:
Work is shifted to Debian and will be targeted as primary.

closed 2017-06-10 by betterclever