fossasia/susi_linux #13: Best Strategy to distribute Susi Hardware

betterclever, 2017-05-31:
Susi Hardware has a dependencies on Libraries like PocketSphinx which takes a long amount of time to install on a Raspberry Pi and similar under- powered development boards (since they are needed to be compiled from source).
We have some potential solutions to save end users from going through the trouble:
- Like Jasper Project (https://jasperproject.github.io/), we can ship the solution as a compressed images with all dependencies.
- Compile a packaged binary for dependencies like PocketSphinx for Raspberry Pi, and distribute it. Users can then install it directly without need to install it from source.

I am looking for better strategies and currently working on fulfilling first.

mariobehling, 2017-06-05:
I think this issue is not about hardware distribution per se. It is actually about the software images. You are raising an important point, but we can also decide this finally later and see what works best for the moment.

betterclever, 2017-06-05:
@mariobehling  Yes, thanks for correcting the title.  It should have been that at the first place. I raised this issue early so that as we proceed along the project as we find better ideas, we may have a list here in the tracker.

mariobehling, 2018-06-17:
@stealthanthrax Let's follow up on this now. How can we get a custom image with SUSI to be installed on an SD Card?

woshikie, 2018-07-09:
For Raspberry Pi, we can use [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/) to copy the partition and settings on a Pi into a `.img` file and we can use the same `.img` file to populate new SD Cards using the same program

stealthanthrax, 2018-07-09:
@woshikie , but first I think we should follow a way (that @Orbiter suggested) to remove x11 libraries from raspbian and make it a minimal environment

woshikie, 2018-07-09:
There's Raspbian Lite

hongquan, 2018-07-09:
We can use Raspbian build script, basing our custom image on Raspbian Lite, adding some modification to let it install some dependencies to the image.

stealthanthrax, 2018-07-11:
@woshikie @hongquan  , there is also an inbuilt SD card copier in Raspbian, I used that everything worked fine. I think that should also be taken into account

stealthanthrax, 2018-08-07:
@hongquan , how can we create a fresh image with just installed software??

hongquan, 2018-08-07:
@stealthanthrax https://github.com/fossasia/susi_linux/issues/323

> Customize Raspbian build script to create image file, including our stuff.

mariobehling, 2018-09-01:
We are following: http://www.yacy-websearch.net/wiki/index.php/En:DebianHeadless

closed 2018-09-01 by mariobehling

hongquan, 2018-09-01:
We are following these steps #349 . I haven't made a detail documentation yet.