fossasia/susi_linux #16: Hardware requirement to make Susi Hardware

betterclever, 2017-06-05:
Our final goal of Susi Hardware is that it should be affordable and cheaper than the alternatives.
The following hardware fulfills the requirement. 
1. [Orange Pi Plus ](https://www.aliexpress.com/item/Orange-Pi-PC-Plus-ubuntu-linux-and-android-mini-PC-Beyond-Raspberry-Pi-2/32668618847.html)  1 GB RAM - **26 USD with Power Supply**
2. [Mini Speakers](https://www.aliexpress.com/item/New-Style-3-5mm-Mini-Balloon-Mobile-Phone-Portable-Small-Sound-External-Put-Mini-Speaker/32706777764.html) - **5 USD**
3. [Mini OLED Display](https://www.aliexpress.com/item/0-96-inch-IIC-SPI-Serial-128X64-Yellow-Blue-OLED-Display-Module-I2C-LCD-Screen-Board/32779485021.html)- **3.71 USD**

Total: **34.71 USD**

mariobehling, 2017-06-05:
What about microphones and housing?

betterclever, 2017-06-05:
@mariobehling, OrangePi contains Microphone built in. No need of external Microphone.
As of casing, such a thing would require someone who has knowledge of 3D printing and is at later stage of the project.
I am not much familiar with it and need some time to get information about it and costs concerned.

Orbiter, 2017-06-05:
There are tons of free OrangePi case designs on Thingieverse, like this one:
https://www.thingiverse.com/thing:1897829

I also have a printer to test a design, I am also able to put together some shapes in 123D, but I am not a 'Designer'. @mariobehling We could use the help of a real designer - for the graphical part.

@betterclever What about https://www.aliexpress.com/item/New-Orange-Pi-Zero-Set-6-Orange-Pi-Zero-512MB-Expansion-Board-Black-Case-development-board/32782429852.html
It is smaller and cheaper but has only 512MB ram. I alreay have that device, could we use it?

betterclever, 2017-06-05:
@Orbiter I cannot strongly approve until I have the actual device. This should be usable as far as I think. One notable point is PyAudio loses Audio Frames frequently on Raspberry Pi 3. I guess it is due to its soundcard but might be due to memory. I am not sure, whether having less memory have some implications on it.
Other than that, I feel no reason it shouldn't work well.

Orbiter, 2017-06-06:
I can do a design maybe, but that is not a blocker here. I think for the hardware we should target something which is small, available and cheap and the current selection is good.

The speaker: I think they are a bit strange. If we make a custom design case, the speaker should be inside the case.

Also: if susi is speaking, the speech recognition should be off, otherwise it could happen that susi addresses itself...

betterclever, 2017-06-06:
@Orbiter Yes, in current prototype above point is being targeted i.e. Recognition is paused until Susi is speaking. 

Also, today I have made good progress with it. Despite some [issues](https://github.com/Uberi/speech_recognition/issues/124) on Speech Recognition Library with raspberry pi, today it is working correctly 3-4 times then hangs.

Also, I made an analysis on memory consumption. On a complete headless system, peak memory usage went to 100 MB when it was running. 
So, we are good even with 512 MB. :smile:

betterclever, 2017-06-06:
Also, about speakers, yeah they are weird :sweat_smile: , I just found the cheapest plug and play speaker. Will find better and compact speakers, which can fit into casing well. :smile:

Orbiter, 2018-05-21:
We are now going for a Raspberry Pi Zero W with a ReSpeaker 2-Mics HAT

closed 2018-05-21 by Orbiter