fossasia/susi_linux #20: Use Flite instead of Festival TTS

betterclever, 2017-06-07:
Currently, we are using Festival TTS. While Festival is excellent for a system with good processing power, it takes considerable amount of time to synthesize speech on English Female Voice being used now on low powered development board like Raspberry Pi.

For example, to synthesize "Hey! I am fine, Thank You" , it takes 10 seconds !!!

We can use [Flite ](http://www.festvox.org/flite/) instead.
Flite stands for Festival Lite and it synthesize US English Female Voice very fast. The above query for example takes mere 1 second to synthesize. 

Thus, we should switch to Flite Speech Synthesis Engine.

mariobehling, 2017-06-08:
I tested Flite. The quality is still not good. Currently we are using Google Speech on Android. Thus, for TTS and STT we are using Google elsewhere anyways. What options are there to use Google here as well?

betterclever, 2017-06-08:
Well, we can use Google TTS, but there is no official way to do so. Google provides no official API for TTS. There is a hack to do it through [Google Translate URL](https://stackoverflow.com/questions/9893175/google-text-to-speech-api/31791632#31791632) and many people use it but if one day Google pulls out support or changes it , it does that in the past once too, we can't do anything.

Secondly, TTS will be done this way on the cloud, which will require additional bandwidth. Google TTS on Android, on the other hand, is completely offline, but being a Platform Specific API, this isn't available on other platforms.

Flite does a good job with correct voice I think. Would you mind seeing a video about interaction and check if it is satisfactory? I shall upload shortly.

Thanks.

mariobehling, 2017-06-08:
Yes, please show the outcome with Flite.

Still, what about other solutions, e.g. IBM Watson? There is a free tier.

betterclever, 2017-06-08:
Yes, IBM Watson can be used for both TTS and STT. They have  best free tier (maximum usage limit) among all other providers. 
btw, Video is almost ready. I'm processing it for upload. Will be done soon. :smile:

mariobehling, 2017-06-09:
@betterclever Can we have a skill for SUSI like "Switch to internal speech recognition" or "Switch to cloud speech recognition"? Please try how it works with Watson. 

Also, what ways are there to speed up the output of Voice? Would better hardware help?

betterclever, 2017-06-09:
The delay in video was majorly due to delayed response from Susi Server in
some queries and slow internet in my home :sad: . Hardware isn't bottleneck
as far as I observe.
Yes, I am trying TTS with IBM Watson.
Also, Speech Recognition shouldn't be done internally at all. Only option
CMU Sphinx, performs very poorly for offline Speech Recognition.
I think, you wished to write about Speech Synthesis (TTS), and for that,
Yes, we can have a skill. :) It won't be hard.

closed 2017-06-14 by betterclever