fossasia/susi_linux #26: Connect Susi Hardware to Susi Webchat for demonstration

betterclever, 2017-06-12:
@Orbiter recommended about connecting Susi Hardware to Susi Webchat so that when we send a request via Speech to our Hardware device, we can view the exact response of Susi in a webchat interface also. This functionality is mainly for demonstration purposes where users can see the exact query.

Exact Requirement as stated on Gitter Channel
[image]

@uday96 @rishiraj824 Can you suggest a possible implementation of it? How can it be achieved?

rishiraj824, 2017-06-12:
@betterclever Yes, we can do this via web sockets. The only thing is the socket needs to be setup, and finally one can listen to that to that socket to listen to the text response of the Hardware.

betterclever, 2017-06-12:
@rishiraj824  Yes, sockets are a good option to achieve this. 
So, I will code up the Python implementation of a websocket interface via which Susi Web Chat and Susi Hardware device can communicate. 
On connecting a monitor and visiting a particular URl or better visiting URL on any device connected to that network, Susi Web Chat interface can be shown which will interact to Hardware device via sockets for getting query input and sending JSON output from server. This way, only one call to server will be made as required. This should be easy. I will report back as soon as it is ready.

rishiraj824, 2017-06-12:
@betterclever :+1:

betterclever, 2017-06-28:
Solved in #30

closed 2017-06-28 by betterclever