fossasia/susi_linux #31: Unable to install on Mac

tstreamDOTh, 2017-06-15:
Unable to install on Mac. I followed the read.me mentioned, and I learned that it is not working for mac os. Please ensure and cross-verify the read.me before publishing it.

betterclever, 2017-06-15:
@tstreamDOTh Our primary development environment is Linux and Readme clearly mentions that. Small computers like Raspberry Pi also primarily run Linux. Thus, instructions are available for Linux only.
And I am afraid, I do not have a Mac currently so cannot provide instructions for the same too. The tools we used might/not support Mac so I am not very sure regarding that. 
README mentions Linux in minimal requirements for now and that is our only aim. If you manage to do it however for mac, feel free to send PR with instructions :smile:

betterclever, 2017-06-16:
Not needed for now.

closed 2017-06-16 by betterclever