fossasia/susi_linux #44: Add support for Home Automation

betterclever, 2017-07-06:
As a goal from my proposal and a needed feature in the Susi Hardware project is support for the Home Automation features in Susi.
Since we focus on Open Source Components, we can add support for openHAB project (https://www.openhab.org/). 
openHAB is an open source Home Automation project and it already has support for Raspberry Pi. The only challenge is to connect openHAB API to correct Susi Action Types.

For example, we can have an ActionType, "turn_on" with "device" argument as "fan" and "location" argument as "living room", so we can call the API of openHAB project for that particular device.

It is just a random example. I am currently reading the documentation of openHAB project and setting it up on my Raspberry Pi. I will write clear working directions in the issue once I am familiar with openHAB API clearly.

mariobehling, 2017-07-07:
This should probably go into the server or be implemented as a skill. It is probably not an implementation that should go into a SUSI client.

betterclever, 2017-07-07:
Yes, I did research on it. openHAB has a myopenHAB service ( http://www.myopenhab.org/ ) and it can connect to home devices via the internet. I was earlier thinking that there is only an Intranet API available and thus the issue.
There is an official Alexa Skill already for it. https://github.com/openhab/openhab-alexa (thought it is yet to be accepted by Amazon)

I will look into the Alexa skill and make SUSI Skill for connecting to myopenHAB service. I will open the follow-up issue in the SUSI skill data repository.

closed 2017-07-15 by betterclever