fossasia/susi_linux #51: SUSI for Meilix

xeon-zolt, 2017-07-21:
we are trying to add SUSI to Meilix and faced this error 
https://github.com/fossasia/meilix/issues/81
I can fix the first 3 errors by editing the /usr/share/alsa/alsa.conf
but how to fix the channel error
[image]

betterclever, 2017-07-22:
I think that error is due to correct device not being used by PyAudio. It seems that you are in a virtualized environments. 
Please check once if Microphone is accessible by recording an audio with ```rec t.wav``` (rec command is in SoX package) and playing it back. 
Also, are you using plain ALSA or PulseAudio/ALSA ? I will also try Meilix on a virtualized environment to make a better config dedicated to Meilix.

mariobehling, 2018-06-22:
Out of scope of the current project.

closed 2018-06-22 by mariobehling