fossasia/susi_linux #55: Add Documentation for detailed Raspberry Pi setup with screenshots

betterclever, 2017-07-31:
Documentation for the Raspberry Pi setup is still inaccurate and a lot of hidden dependencies are being missed which users encounter at time of installation.

All such dependencies are need to be installed earlier and specified so that installation is straight forward. For this, a guide with proper screenshots for each step and correct commands must be added.

mariobehling, 2017-07-31:
There should be rather a package that calls all the necessary dependencies instead of requesting users to install packages themselves.

betterclever, 2017-07-31:
I am adding all the dependencies to the installation script so manual installation won't be needed. Users just need to run the script once.

PR #56 (2017-08-01): Added proper setup instructions for Raspberry Pi and Ubuntu
Fixes #55 

**Changes**
- Added proper setup instructions for Raspberry Pi (Tested twice on Raspberry Pi 3)
- Added proper setup instructions for Ubuntu (Tested on fresh installation of Ubuntu 16.04 and Ubuntu 17.04 on x64 chipset)
closed 2017-08-01 by mariobehling