fossasia/susi_linux #85: Feature to properly quit from the system.

isht3, 2017-09-15:
To exit from the system, we have to `crtl+c` while running, which is not clean. We can discuss better alternatives here and I'll get on with it implementing it right away.

I was thinking maybe pressing `q` while the process was running to quit?

isht3, 2017-09-15:
@betterclever Suggestions? :smile:

betterclever, 2017-09-15:
To quit the system properly, we need to force system to idle state. Then, send interrupt signals to speed detection loop and stop the hotword detection. For CLI, we may do it using 'q' button as you suggested. For GUI mode, make sure it happens atomically on close button press.

isht3, 2017-09-16:
@betterclever I will get right on with it. Thanks for explaining!