fossasia/susi_linux #94: What is the status of MMM-SUSI-AI?

E3V3A, 2018-03-11:
Hi Guys! 

We're a bunch of developers for [MagicMirror](https://github.com/MichMich/MagicMirror)and we would like to use SUSI.AI. We asked some questions int he [MMM-SUSI-AI](https://github.com/fossasia/MMM-SUSI-AI) repo, but it seem abandoned...  :broken_heart: 

Can you please let us know which sub-repository of SUSI.AI we should pay attention to in order to get this working on our mirrors?

Thanks in advance,
:peace_symbol: and :heart: 
E:V:A

betterclever, 2018-03-11:
Hey! I was a little busy couldn't reply to that issue you raised on the MMM-SUSI-AI repo. I created it around July, it used to work then but I guess there is some dependency issue now. I'll check the current status and fix that soon. Did you try installing it using the given instructions? If it failed, can you give error logs. That would be helpful.

E3V3A, 2018-03-12:
Hi @betterclever ! 
Thanks for getting back to us. We haven't tried to install it yet, because:
1. We don't know if it is still mainatined?
2. We don't know if **this** repo development has superseeded the MMM,
3. What are the dependencies and requirements to use the MMM?
4. What services (API) does it use?

betterclever, 2018-03-12:
This repo is independent of MMM-SUSI-AI. That is being maintained though I
didn't get the time in a while to update packages in a while.