fossasia/susi_linux #97: Syntax error during main file execution

Donyme, 2018-03-18:
**Actual Behaviour**
[image]
Syntax error

**Expected Behaviour**
No syntax error

Steps to produce it :
run the command python3 -m main in the susi_linux folder

prateekiiest, 2018-05-18:
This will work with python 3 not python 2

hongquan, 2018-05-19:
Your screenshot shows that you ran with Python 2, not 3.

And the line of code

```python
print('%s\t' % h, end='')
```

is actually for Python3, not 2.

In Python 2, `print` is language statement, not function. So, that line, if run in Python 2, will be equivalent to

```python
print ('%s\t' % h, end='')
```

where `('%s\t' % h, end='')` is a tuple. But in a tuple, there cannot be "keyword" syntax `end=''`, so `SyntaxError` is raised.

Donyme, 2018-05-22:
Thank you! For the explantion!

closed 2018-05-22 by Donyme