fossasia/susi_linux #105: Error during execution of "config_generator.py" file.

Santan47, 2018-05-15:
[image]
no module named as json_config.

stealthanthrax, 2018-05-15:
@Santan47 , did you execute install.sh file??

Santan47, 2018-05-16:
Hi @stealthanthrax , yes I executed the install script in susi_linux folder.

[image]

I am using ubuntu 14.04 OS and after execution of install.sh  I got an output like this.
There are some packages which are not installed in the execution and showing me error.
so can you tell me how can i resolve this error.

hongquan, 2018-05-17:
You are using too old Ubuntu (14.04), which doesn't have suitable software.

prateekiiest, 2018-06-21:
is this issue resolved, may be we can close this

closed 2018-06-25 by stealthanthrax