fossasia/susi_linux #111: Deploying a local server on the raspberry Pi

stealthanthrax, 2018-05-17:
**Actual Behaviour**

Only online server available in the wrapper

**Expected Behaviour**

To configure it with a local server

**LogCat for the issue**

Right now , while deploying the server on the raspberry pi , I am getting an error:
````
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
````

**Would you like to work on the issue?**

First, I'll fix the above issue and the continue with the deployment

betterclever, 2018-05-18:
@hongquan The Susi Linux client is written in Python, the actual Susi server runs on Java. The issue is about deploying full Susi Server on the Client.

stealthanthrax, 2018-06-10:
Fixed Here : [link](https://github.com/fossasia/susi_linux/pull/118)

closed 2018-06-10 by stealthanthrax