fossasia/susi_linux #116: Add support for "stop" action in SUSI response

imjog, 2018-05-19:
**Actual Behaviour**

Currently, new 'stop' action is introduced but the action is not implemented.

**Expected Behaviour**

With `stop` input SUSI should stop anything it's doing at that time. 'stop' action is introduced to enable user to make SUSI stop anything it's doing.
Try to make SUSI respond to “Stop” (e.g. SUSI is narrating a story and if the user says STOP, it should stop narrating the story).

**Steps to reproduce it**

Ask SUSI _stop_, you will get the following actions object:
```
"actions": [{"type": "stop"}],
```
Helpful Links: [skill_link
](https://github.com/fossasia/susi_server/blob/development/conf/susi/en_0001_foundation.txt#L41)

JSON Response:
```
{
		query: "stop",
		count: 1,
 		client_id: "aG9zdF8xNzIuNjguMTQ0LjU3X2Q5NzI1MGMx",
 		query_date: "2018-05-19T22:18:48.641Z",
 		answers:  [
     		 {
         		data:  [
             		 {
                 		0: "stop",
                 		actions:  [
                     		 {
                         		type: "stop"
                     		}
                 		],
                 		token_original: "stop",
                 		token_canonical: "stop",
                 		token_categorized: "end",
                 		timezoneOffset: "-330",
                 		language: "en",
                 		skill_link: "https://github.com/fossasia/susi_server/blob/development/conf/susi/en_0001_foundation.txt",
                 		query: "stop",
                 		skill: "/susi_server/conf/susi/en_0001_foundation.txt"
             		},
             		 {
                 		skill_link: "https://github.com/fossasia/susi_server/blob/development/conf/susi/en_0001_foundation.txt",
                 		query: "stop",
                 		skill: "/susi_server/conf/susi/en_0001_foundation.txt"
             		},
             		 {
                 		skill_link: "https://github.com/fossasia/susi_server/blob/development/conf/susi/en_0001_foundation.txt",
                 		query: "stop",
                 		skill: "/susi_server/conf/susi/en_0001_foundation.txt"
             		},
             		 {
                 		skill_link: "https://github.com/fossasia/susi_server/blob/development/conf/susi/en_0700_ai_play.json",
                 		query: "hey susi",
                 		skill: "/susi_server/conf/susi/en_0700_ai_play.json"
             		}
         		],
         		metadata:  {
             		count: 4
         		},
         		actions:  [
             		 {
                 		type: "stop"
             		}
         		],
         		skills:  [
             		"/susi_server/conf/susi/en_0001_foundation.txt"
         		],
         		persona: { }
     		}
 		],
 		answer_date: "2018-05-19T22:18:48.646Z",
 		answer_time: 5,
 		language: "en",
 		session:  {
     		identity:  {
         		type: "host",
         		name: "172.68.144.57_d97250c1",
         		anonymous: true
     		}
 		}
}
```

**Screenshots of the issue**

N/A

**Would you like to work on the issue?**

No

stealthanthrax, 2018-05-20:
Thanks @imjog , I'll try to implement it susi_linux

stealthanthrax, 2018-05-27:
This issue is fixed here https://github.com/fossasia/susi_api_wrapper/pull/39

prateekiiest, 2018-07-01:
Can this be closed?

imjog, 2018-07-01:
@prateekiiest @mariobehling yes, it can be closed since it is implemented on all the clients.

stealthanthrax, 2018-07-01:
@imjog , this shouldn’t be closed right now, as there is a dependent bug whose PR is in progress

PR #232 (2018-07-07): Added stop action in the repo
Fixes #116 

This PR implements stop action in SUSI linux repo and creates a base for #204
closed 2018-07-07 by Orbiter