fossasia/susi_linux #121: Play audio from a Youtube stream

Orbiter, 2018-05-21:
**Expected Behaviour**

When susi returns a audio_play action, the hardware which the user is using must play the given audio source.
This is a follow-up feature to https://github.com/fossasia/susi_linux/issues/117

**Steps to reproduce it**

Ask susi "open the pod bay door" - https://api.susi.ai/susi/chat.json?q=open+the+pod+bay+door
You will get the following action object:
```
"actions": [
      {
        "type": "audio_play",
        "identifier_type": "youtube",
        "identifier": "7qnd-hdmgfk"
      },
      {
        "language": "en",
        "type": "answer",
        "expression": "I'm sorry, Dave. I'm afraid I can't do that."
      }
    ]
```
This means that susi shall do (in that order):
* play the audio from the youtube video with the given ID. Use the "identifier_type" to see that the "identifier" object denotes a youtube ID
* do not use a environment which requires a X11 library. There should be linux tools to grab and play audio from a youtube video.
* write out the answer or speak it (according to the current interface)

In text-based interfaces the writing out makes sense, in headless clients this may be disturbing so we might want to find out how to handle this. However just output both for now. We will have a further issue to make this better

**Documentation of the audio_play action**

see issue https://github.com/fossasia/susi_linux/issues/117

**Would you like to work on the issue?**

Please let us know if you can work on it or the issue should be assigned to someone else.

stealthanthrax, 2018-05-22:
@Orbiter , can I use this http://tizonia.org/ to play music?

betterclever, 2018-05-22:
I will recommend checking out
https://github.com/anyaudio/anyaudio-server

It is a completely open source project made by a long term FOSSASIA
contributor Avi Aryan.

stealthanthrax, 2018-05-23:
@betterclever , does it consist of any x11 libraries?

betterclever, 2018-05-23:
It should not. It is meant to be a cross-platform app.

stealthanthrax, 2018-05-23:
@Orbiter , the issue has been solved here https://github.com/fossasia/susi_api_wrapper/pull/37

stealthanthrax, 2018-05-28:
Fixed here : https://github.com/fossasia/susi_api_wrapper/pull/37

stealthanthrax, 2018-07-11:
It is fixed and hence can be closed

closed 2018-07-11 by stealthanthrax