fossasia/susi_linux #126: Create Media Discovery Daemon

Orbiter, 2018-05-26:
**New Behaviour**

Susi in local mode shall discover media from a USB device.
A new daemon must watch USB activities and check if media is inserted.
If new media is discovered, the daemon writes a file containing Intents in Susi Skill language which can be used to play the media. These intents must have simple audio_play actions and an URL which points to a file:// location in the local file system

The storage place for these skill files is `<susi_home>/data/generic_skills/media_discovery`
Susi will watch this path and discover any file placed there automatically.

One of these skills must contain an intent phrase like "play from USB" or "play USB". All other files should then playable with "next". The auto-generated file must cover this functionality.

Orbiter, 2018-05-26:
To test this: the local susi_server must have the property `local.mode = true`, otherwise the media_discovers path is not watched.

stealthanthrax, 2018-05-27:
@Orbiter , should the daemon be running always or should it be triggered by some command?

hongquan, 2018-05-28:
If your desired feature is just to continuously watch and detect if USB device is plugged in, then execute some command, it is more naturally to use [udev](https://linuxconfig.org/tutorial-on-how-to-write-basic-udev-rules-in-linux).

stealthanthrax, 2018-05-28:
@hongquan , it will be easier to write susi_skills from this python script when the USB will be detected

Orbiter, 2018-05-29:
If that works, yes. Same must be the other way if it is detected that the USB is disconnected.

stealthanthrax, 2018-07-06:
Fixed here : [link](https://github.com/fossasia/susi_linux/pull/157)

closed 2018-07-06 by stealthanthrax