fossasia/susi_linux #135: To add a hardware button that will trigger the factory reset process

stealthanthrax, 2018-05-31:
**Actual Behaviour**

Right now the factory reset process can be triggered only by the terminal

**Expected Behaviour**

To use hardware switch to allow factory reset

**Would you like to work on the issue?**

Yes

mariobehling, 2018-06-22:
Here is an approach mentioned by @hongquan 

> Divide storage (flash) to 2 partitions: Read-Only one (1) and Writable (2).
Programs are placed in 1st partitions. Their default configuration are also placed in 1st partition. Changed configuration are placed in 2nd partition. They then use some technique (called "overlay") to place the configuration of the 2nd partition to be on of top 1st partition. For example: they have the network config at /etc/config/network. But if user change it (change SSID & password), the new content are stored in 2nd partition and later access of this file will get the content from the 2nd partition, not the 1st.
They reserve a hardware button for reseting, connected to a GPIO port. If this button is pressed, a program is triggered, which we erase the content of 2nd parition: All configuration are reset to default, corresponding to what stored in partition 1.

hongquan, 2018-06-22:
Now, I think we cannot follow totally OpenWrt's approach, because:

- Our software is just an app, not an OS. We cannot define how partition should be.
- Changing partition layout will involving modify Raspbian build script. If we do too much modification, we will end up maintaining a new OS!

So, instead of dividing to 2 partitions (read-only and writable), we can divide our app to 2 parts:
- Software body, saved in root-owned folder (_/usr/local/_ or _/opt_), so that it cannot be overwritten.
- Configuration, saved in user's home directory, or _/etc/_.

For resetting, we just delete the configuration files.

stealthanthrax, 2018-07-09:
@hongquan , we already make a config.json file , we can delete just that and uninstall and install all the packages. This method would contain it in a single script

hongquan, 2018-07-10:
But we also need the body of application to stay read-only, to prevent overwriting by accident.