fossasia/susi_linux #164: Media discovery daemon shall write susi skills

Orbiter, 2018-06-15:
**Expected Behaviour**

If the user plugs in a USB drive, the discovery daemon reads the USB drive and writes a skill file which can be used to play the media.

How it works:
- a local server is running on the device. It is looking for a skill file in the path `data/generic_skills/media_discovery/`
- a skill in that path must have intents (at least one, maybe more) to please each of the audio. The intent would look similar to the example skill
```
system check
!console:all systems are go, all lights are green!
{"actions":[
{"type":"audio_play", "identifier_type":"url", "identifier":"file://conf/audio/all_systems_are_go_all_lights_are_green.mp3"}
]}
eol
```

To make this working, a local susi must set in the data/settings/customized_config.properties the following properties:
```
local.mode = true
```

**Would you like to work on the issue?**

Please let us know if you can work on it or the issue should be assigned to someone else.

Orbiter, 2018-06-15:
in case that the media is removed from USB, the skill file must be removed again.

Orbiter, 2018-06-15:
The name of the skill file may be {any}.txt

stealthanthrax, 2018-06-20:
Fixed here : [link](https://github.com/fossasia/susi_linux/pull/159)

closed 2018-06-20 by stealthanthrax