fossasia/susi_linux #172: To document the workflow of connection between Smart speaker and IOS/Android App

stealthanthrax, 2018-06-18:
**Actual Behaviour**

No documentation present

**Expected Behaviour**

To add documentation

**Would you like to work on the issue?**

Yes

Orbiter, 2018-06-23:
[image]

I tried to draw the process. Please discuss.

mariobehling, 2018-06-23:
Yes, this is the plan.

Orbiter, 2018-06-23:
When the Susi Device is creating it's own hot spot, we should use a fixed name for that so its easier to identify that the Android/IOs device has actually found the right one. Lets call the ssid "susi.ai", that would create some advertisement in the very moment when the hotspot is on.

mariobehling, 2018-06-24:
Yes, please use it in caps "SUSI.AI".

mariobehling, 2018-06-28:
Unfortunately SVGs don't display on github. Please add the svg code into an independent file in the folder `docs/images` and use "rawgit.com" to add it into the Readme.md.

PR #202 (2018-06-28): Added SUSI Smart Speaker Workflow
Fixes #172

Changes
Added the SVG for the workflow

Screenshots for the changes
[image]
PR #213 (2018-06-30): Documented Media Daemon Workflow
Fixes #172 

Changes

Added Media Daemon workflow and used SVG images hosted on rawgit rather than inline SVG

Screenshot

[image]
closed 2018-06-30 by mariobehling