fossasia/susi_linux #182: To add a script to configure other scripts on work on bootup

stealthanthrax, 2018-06-20:
**Actual Behaviour**

Many present scripts require to executed on boot-up. 

**Expected Behaviour**

To create a script that will allow the other script to execute on startup

**Would you like to work on the issue?**

Yes

stealthanthrax, 2018-07-06:
@hongquan , the user will have to implemented manually, or I can add it in `install.sh`

hongquan, 2018-07-07:
All the scripts that need to be started on bootup have to be implemented as `systemd` service.
The _install.sh_ should take care of installing those services.

hongquan, 2018-07-07:
Later on, let's create a folder named _Deploy_ to contains files used for deployment. The structure can be:

```
+ Deploy/
    + Systemd/
          +-- ss-ap-server.service
          +-- ss-update-daemon.service
    + Others...
```

The systemd service files are prefixed `ss`, means `susi`, to mark that they are from our project.
When installing SUSI.AI, the _ss-*.service_ files are then copied to _/lib/systemd/system/_. Note, don't copy to _/etc/systemd_, because the former location allow us to _disable/enable_ services when needed.