fossasia/susi_linux #195: To decide the required digital ampli

stealthanthrax, 2018-06-25:
**Actual Behaviour**

No digital amplifier is being used

**Expected Behaviour**

To decide an external digital amplifier required

**Would you like to work on the issue?**

Yes

woshikie, 2018-06-28:
I was thinking about [https://www.adafruit.com/product/1712](this i2c amplifier from Adafruit)

cweitat, 2018-06-30:
https://www.adafruit.com/product/3006
Looking at this as alternative as well

woshikie, 2018-07-06:
https://github.com/respeaker/seeed-voicecard/issues/39
looks like the re-speaker already has a built-in speaker. I am going to test the loudness

woshikie, 2018-07-06:
Tested both 2W8Ω and 1W8Ω speakers by connecting them raw (no additional amplifier) to the respeaker's Speaker Out port by soldering two extra jumper cables to the back of the JST 2.0 Male Port.

Both speakers works well and tested to be able to hear at about 5 metres away. The microphone is also picking up echos from the speaker at the same distance.

woshikie, 2018-07-16:
https://www.robot-r-us.com/components/digital-potentiometer-10k.html

mariobehling, 2018-09-01:
We are sticking with Respeaker 2 Head for now.

closed 2018-09-01 by mariobehling