fossasia/susi_linux #204: SUSI stops listening while processing the queries

stealthanthrax, 2018-06-27:
**Actual Behaviour**

SUSI stops listening while processing the queries, so actions like stop action do not work

**Expected Behaviour**

To fix this bug

**Steps to reproduce it**

Try giving some command while SUSI is processing a previous query

**Would you like to work on the issue?**

Yes

stealthanthrax, 2018-07-06:
@Orbiter @hongquan @cweitat @woshikie , can you give some suggestions on this issue , as this is causing a bit of the blocker

hongquan, 2018-07-23:
I think, instead of running `sox`, putting it to the background with shell, then use "kill" to stop the music, we should use [Music Player Daemon](https://en.wikipedia.org/wiki/Music_Player_Daemon) to running in the background as daemon, and request it to play/stop music file.

stealthanthrax, 2018-07-23:
@hongquan , but right now we majorly use mpv to play youtube videos . And not a lot of sox player. But we can try using mpd to play usb audio.

hongquan, 2018-07-23:
The PR #292 is a bug. Reopen this issue now.

stealthanthrax, 2018-07-23:
@hongquan , it works without quitting anything . And yes we need to look for a better way. I am looking on it

hongquan, 2018-07-23:
It works now, doesn't mean that it works later, when the bug is still there. The bug will make it not working in the future!

hongquan, 2018-07-24:
@stealthanthrax 
>  but right now we majorly use mpv to play youtube videos

Sorry, I misunderstood.

So, most of `sox` usage here are to play short audio file, it won't block our app.
The issue is about Youtube video, which is long and blocks.

Ok. Then:

- Check if `mpv` can work as daemon (like MPD).
- If not, run it in a separate thread. Our app become asynchronous now. Should design a communication channel between main thread and `mpv` thread, so that, when we want to stop the song, we can send a command from main thread to "mpv" thread to ask it to make `mpv` stop.

PR #292 (2018-07-23): Susi Now processes two queries while playing Music
Partial Fix #204 

Now, while SUSI is playing music, you can use the hotword "SUSI" to activate SUSI and the give another query.

Next Steps:
To be able to turn down the music while SUSI is processing the second query and to be able to process "Second Query", while SUSI is giving out normal answers