fossasia/susi_linux #206: More Installation Errors

woshikie, 2018-06-28:
**Describe the bug**
Installation Errors

**To Reproduce**
Steps to reproduce the behavior:
1. Flash a new version of raspbian to raspberry pi
2. Try to install susi on it 

**Expected behavior**
A smooth installation without errors

**Screenshots**
[Terminal Log(Pastebin)](https://pastebin.com/GdWWNHBh)

**Additional context**
A different error occurs on ubuntu 18.04

stealthanthrax, 2018-06-28:
@woshikie , would you like to work on these issues, or can these be assigned to someone else?

woshikie, 2018-06-29:
@stealthanthrax I'd love to, but I have some questions.
1) why is graddle needed? Isn't susi a python app? 
2) why is the local server started automatically? 
3) i ran into some issues after this, can you help?

stealthanthrax, 2018-06-29:
@woshikie ,
1> Our main SUSI server is made in JAVA hence we need a gradle to build and deploy it locally.
2> The local server is started automatically since we are using the local server as the default, so we'll try and connect to it every time
3> I can try and solve issue #207 ??

stealthanthrax, 2018-06-30:
@woshikie , the local server increases the efficiency by around 20% and if there is a failure in local server : say a build failure , we switch to the online server

woshikie, 2018-06-30:
@stealthanthrax and what should happen if there's a build failure for the local server and the online server is offline for some reasons?

stealthanthrax, 2018-06-30:
@woshikie , the user can do a soft reset to start the build again

woshikie, 2018-06-30:
@stealthanthrax if the user is unable to build it the first time, how is it possible that the use is able to build it even after soft reseting?

woshikie, 2018-07-02:
Reopened because issue is still happening

stealthanthrax, 2018-07-02:
@woshikie , i am unable to replicate this bug. Please can you elaborate?

woshikie, 2018-07-02:
@stealthanthrax i have updated the log

stealthanthrax, 2018-07-05:
@woshikie , there are some updates made now , can you check again on the pi 0 ??

woshikie, 2018-07-06:
@stealthanthrax gradle build still failed at 

- GeocodingTest > testNoTFound
- GeocodingTest > testGeocoding

woshikie, 2018-07-06:
@stealthanthrax i have updated the log on the first post, please take a look

stealthanthrax, 2018-07-06:
@woshikie , this a problem with pi zero's architecture and JVM version. I think this can be solved only from the server side repo

woshikie, 2018-07-07:
workaround found for raspberry pi zero w

woshikie, 2018-07-07:
Looks like java (oracle) that came with the raspberry pi does not support server mode on armv6.
To fix that, we have to install openjdk java8 (be careful not to install default-jdk because it will download the latest openjdk java, which is version 9. Java has dropped support for armv5/6/7 in java 9)

PR #212 (2018-07-01): Fixed a minor bug while installation
Fixes #206 

Changes
This fixes the bug which was not allowing the installation of dependent libraries if there was some issue in Server build
PR #241 (2018-07-09): Updated installation steps for raspberry pi
Added extra steps needed for armv6 devices such as raspberry pi zero/ zero w/ zero wh

Fixes #206
closed 2018-07-09 by woshikie