fossasia/susi_linux #207: ImportError when running config_generator.py

woshikie, 2018-06-28:
**Describe the bug**
ImportError occured when running config_generator.py

**To Reproduce**
Steps to reproduce the behavior:
1. Flash a fresh version of Raspbian Image
2. Install SUSI on it, expect errors.
3. Run `python3 config_generator.py`

**Expected behavior**
No errors and completes the configuration

**Screenshots**
```
pi@raspberrypi:~/susi_linux $ python3 config_generator.py
Traceback (most recent call last):
  File "config_generator.py", line 4, in <module>
    import json_config
ImportError: No module named 'json_config'
```

stealthanthrax, 2018-06-28:
@woshikie , you have to run `install.sh` file

woshikie, 2018-06-29:
@stealthanthrax I already did

hongquan, 2018-07-05:
The `json_config` seems to be installed somewhere by the _install.sh_, but this _install.sh_ failed on your machine, so you didn't get `json_config`. You have to install it manually.

PR #225 (2018-07-05): Fixes the errors in config_generator.py file
Fixes #207 

Fixes the minor errors in config_generator.py file
closed 2018-07-05 by hongquan