fossasia/susi_linux #215: Make an automated support for using either Snowboy or Pocketsphinx when one is not installed.

prateekiiest, 2018-07-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Currently under `hotword`, we let the user decide if they have Snowboy installed on their device and ask them to use it if they have (due to its much more features). On the other hand if Snowboy is not installed we ask them to go for PocketSphinx and use it.  

Instead of this we can easily do it through the code for the user by checking if the corresponding library is available without the user having to bother which hotword detection to import.

**Describe the solution you'd like**
Make a `try catch ` exception while importing the hotword detection library under https://github.com/fossasia/susi_linux/blob/master/main/hotword_engine/__init__.py

https://github.com/fossasia/susi_linux/blob/912948bf34eb43018f71fde44e6ff5ea44eaab2a/main/hotword_engine/__init__.py#L10

A clear and concise description of what you want to happen.

So we can see if we get an `ImportError` while importing snowboy, then we can import the Pocketsphinx and vice-versa. This will not let the user to bother about which library to used without having him/her to check which one is installed. It would simply try importing the libraries and if exception occurs in one it will go for importing the other module
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
NA
**Additional context**
Add any other context or screenshots about the feature request here.
NA

I would like to work on it

prateekiiest, 2018-07-02:
@stealthanthrax I am unable to add labels without right access. Could you assign me this and add appropriate labels for this issue?

stealthanthrax, 2018-07-02:
@prateekiiest , what if the user wants to user Pocketsphinx by default??

prateekiiest, 2018-07-03:
you are saying when both the libraries are installed properly?

we make an exception only when snowboy is not present, then use Pocketsphinx. Else in all other cases use Snowboy ?

stealthanthrax, 2018-07-03:
Right now we provide the user a choice to select from the library, should we continue to do it? @mariobehling @Orbiter @hongquan @cweitat @woshikie . What is opinion on this??

prateekiiest, 2018-07-03:
So under `init.py`  we have

```
from .snowboy_detector import SnowboyDetector
from .sphinx_detector import PocketSphinxDetector
```

instead of that

```
try:
    from .snowboy_detector import SnowboyDetector

except ImportError:
       print('Snowboy Detector is not present')
       from .sphinx_detector import PocketSphinxDetector

try: 
    from .sphinx_detector import PocketSphinxDetector
except ImportError:
   print('PocketSphinx not installed')
   print('We recommend using Snowboy')
   from .snowboy_detector import SnowboyDetector
     

```

   
Some thing like this, without the user to choose.

prateekiiest, 2018-07-04:
@hongquan thanks for your input on this. Can you please elaborate how we can detect that using `import`?
Does the above approach that I mentioned look plausible?

stealthanthrax, 2018-07-04:
@prateekiiest , instead of using `from .sphinx_detector` use `from sphinx_detector`

hongquan, 2018-07-05:
@prateekiiest 

```py
SNOWBOY_AVAILABLE = False
POCKETSPHINX_AVAILABLE = False
try:
	from snowboy.snowboydetect import SnowboyDetect
	SNOWBOY_AVAILABLE = True
except ImportError:
	pass
```

Note, the symbol looks different (`snowboy.snowboydetect.SnowboyDetect` vs `snowboy_detector.SnowboyDetector`) because mine is using standard installation of snowboy.

hongquan, 2018-07-05:
Looking into _hotwork_engine_ folder:

[image]

The `snowboy_detector` module is the local module of `hotwork_engine`, so that you have to import by `from .snowboy_detector import ...` in *\_\_init\_\_.py* script.

Inside _snowboy_detector.py_:

```py
import os

from .hotword_detector import HotwordDetector
from .snowboy import snowboydecoder

TOP_DIR = os.path.dirname(os.path.abspath(__file__))
RESOURCE_FILE = os.path.join(TOP_DIR, "snowboy/resources/susi.pmdl")

class SnowboyDetector(HotwordDetector):
```

It is trying to import `snowboy` from the local folder (as seen in the screenshot). So, to use the standard installation of _snowboy_, you have to delete the _snowboy_ folder, and fix the import line in _snowboy_detector.py_.

prateekiiest, 2018-07-05:
Thanks @hongquan for giving a clear view on this. 
I have sent a PR for deleting the snowboy folder.

hongquan, 2018-07-06:
@prateekiiest Which PR is to delete Snowboy? We have 2 things to do:
- Delete local _snowboy_ folder.
- Install _snowboy_ by standard method (`pip3 install` or `python3 setup.py install`)
- Fix the `hotword_engine_ to let it use standard `snowboy`. Currently, it is relying on the local `snowboy`.

prateekiiest, 2018-07-06:
@hongquan I am only currently deleting the `cp` operation for moving files.
PR #231 

I will fix the third point under #228

prateekiiest, 2018-07-12:
Currently the things that has been done are follows

- [x] Delete local snowboy folder. https://github.com/fossasia/susi_linux/pull/228/commits/3a9ba4a2d124ea3f43ae664f9e8eb68586e00720
- [x] Install snowboy by standard method (pip3 install or python3 setup.py install) 
https://github.com/fossasia/susi_linux/pull/231/commits/8446e79150f1863a30d2c263ac886e1d03b50360
- [ ] Fix the hotword_engine to let it use standard snowboy. Currently, it is relying on the local snowboy.

I have already merged the patch-2 changes of PR https://github.com/fossasia/susi_linux/pull/231 into PR https://github.com/fossasia/susi_linux/pull/228

PR #228 (2018-07-13): Use either of the hotword engine when one is not properly installed
- [x] Delete Local Snowboy folder
- [x] Fix import in snowboy_detector
- [x] Make necessary changes under __init_.py
- [x] Dont copy snowboy local folder
 
Fixes #215 .

We need to decide if this can be a preferable approach
commit 87d4495b (2018-07-13): Merge pull request #228 from prateekiiest/patch-1

Use either of the hotword engine when one is not properly installed.

This is to fix #215
closed 2018-07-13 by hongquan