fossasia/susi_linux #219: Fix installation process

hongquan, 2018-07-03:
This issue is to collect some fix need to make in the installation script.

The target is to make the installation work on Raspbian Stretch (9) and Ubuntu 18.04, and make the process simpler.

Some core idea:
- Try to install the dependency software from the OS repository (via `apt`) and from Python [PyPI](https://pypi.org/) (via `pip3`) first. Only install from source (`make`) if the previous method fails.

hongquan, 2018-07-03:
- SWIG should be installed from Debian repository.
- `libatlas-dev` doesn't exist. Install `libatlas-base-dev` instead.

hongquan, 2018-07-03:
- Tizonia default installtion script only installs Python 2 packages (from https://dl.bintray.com/tizonia/raspbian), not Py3. Doesn't hae a fix yet!

stealthanthrax, 2018-07-03:
* The final step in `install.sh` doesn't prompt the user to use `python3 config_generator.py` however, the new format is 'python3 config_generator.py stt tts hotword wake'

hongquan, 2018-07-03:
From @stealthanthrax feedback, we use Tizonia just for playing Youtube, but Tizonia is buggy, support too many services, and doesn't support Python 3, so we should find a replacement.

hongquan, 2018-07-03:
- Flite can be installed via `apt`.
- The udev rule in _media_daemon/media_daemon.sh_ looks weird:

```
echo "ACTION==\"add\", RUN=\"/home/pi/SUSI.AI/susi_linux/media_daemon/autostart.sh\"" >> 99-com.rules       
echo "ACTION==\"remove\", RUN=\"/home/pi/SUSI.AI/susi_linux/media_daemon/autostop.sh\"" >> 99-com.rules 
```

It is supposed to tell _udev_ to run the _media_daemon/autostart.sh_ script when USB drive is plugged in, and run the _media_daemon/autostop.sh_ when USB drive is removed. But the full rule

```udev
ACTION=="add", RUN="/home/pi/SUSI.AI/susi_linux/media_daemon/autostart.sh"
```
doesn't tell `udev` to only look into USB devices!

By the way, after writing these lines of rules, my Pi's wifi stop working (don't know if it is broken by this udev rules or not).

- The `99-com.rules` file is of Raspbian. We must not write our rule to it. Write to new file, so that when we uninstall the app, we just delete our file.

hongquan, 2018-07-03:
- When I plug my USB drive in, then run `python3 auto_skill.py` manually, the script doesn't detect my drive.

```sh
$ python3 auto_skills.py                                                                        
[]
<class 'list'>
<class 'list'>
Traceback (most recent call last):
  File "auto_skills.py", line 51, in <module>
    make_skill()
  File "auto_skills.py", line 9, in make_skill
    print(name_of_usb[0])
IndexError: list index out of range
```
while `lsblk` still shows my drive:

```sh
$ lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    1  7.5G  0 disk
└─sda1        8:1    1  7.5G  0 part
mmcblk0     179:0    0 29.7G  0 disk
├─mmcblk0p1 179:1    0 43.1M  0 part /boot
└─mmcblk0p2 179:2    0 29.7G  0 part /
```

hongquan, 2018-07-03:
Some Python packages, like [snowboy](https://github.com/Kitt-AI/snowboy.git), has to be installed from Git source, just because the version published to [PyPI](https://pypi.org/project/snowboy/) is too old.

To ease the process, I think one of us can make the wheel package of the newer version and upload to some PyPI-like host, so other can install quickly via `pip3`. And the advantage of wheel package is that, it is prebuilt binary, the next user doesn't need to build again and also doesn't need to install library header packages (_libxxx-dev_).

This is example, when I built snowboy from Git source to wheel package, hosting it [somewhere](https://repo.fury.io/agriconnect/), and let `pip` download from that host when installing snowboy in other machine.

```
$ sudo pip3 install snowboy
Collecting snowboy
  Downloading https://repo.fury.io/agriconnect/-/647580/snowboy-1.3.0-py3-none-any.whl (12.3MB)
    100% |████████████████████████████████| 12.3MB 21kB/s
Requirement already satisfied: PyAudio in /usr/local/lib/python3.5/dist-packages (from snowboy)
Installing collected packages: snowboy
Successfully installed snowboy-1.3.0
```

In the _install.sh_ script, we are trying to copy snowboy to _susi_linux_ folder:

```sh
if [ -f _snowboydetect.so ]; then
	echo "Moving files"
	cp _snowboydetect.so ${root_dir}/main/hotword_engine/snowboy
	cp snowboydetect.py ${root_dir}/main/hotword_engine/snowboy
```

I think this practice is not good. We should let the 3rd-party software live in its standard folder.

stealthanthrax, 2018-07-03:
@hongquan , the autoskills.py script is working fine on my side , please try using a different USB or the same USB with the different name

hongquan, 2018-07-04:
@stealthanthrax Found the cause: Your script tries to call `mount` command, which requires _root/sudo_ permission.
But our app, as every general application, should not be run under `root`, or `sudo`.

hongquan, 2018-07-05:
More discussion about "Media Discovery not working" are in #223.

More discussion about "Media Discovery breaks wifi" are in #226.

prateekiiest, 2018-07-05:
@hongquan are you working on  https://github.com/fossasia/susi_linux/issues/219#issuecomment-402227526 to keep the 3rd party software in its required folder ?

May be I can send a PR for you?

hongquan, 2018-07-05:
@prateekiiest No, I'm not working on it.

prateekiiest, 2018-07-05:
Then I will open an issue for it and send a PR accordingly

hongquan, 2018-07-06:
Regarding `snowboy` issue, we have 3 things to do:
- Delete local _snowboy_ folder.
- Install _snowboy_ by standard method (`pip3 install` or `python3 setup.py install`)
- Fix the `hotword_engine` to let it use standard `snowboy`. Currently, it is relying on the local `snowboy`.

It will be good if you can combine these to 1 PR, so that it is not half-cooked and break the system.

prateekiiest, 2018-07-06:
https://github.com/fossasia/susi_linux/issues/215#issuecomment-402923779

stealthanthrax, 2018-07-06:
@hongquan , one instance of swig is already being installed form the debian repos , the one being installed locally is used to compile the python wrapper of snowboy

stealthanthrax, 2018-07-06:
@hongquan , swig versions are being installed from source as well as the debian repo's. Should I delete the one from the source just like that??

hongquan, 2018-07-07:
We need to convert susi_api_wrapper to setuptools-based package (https://github.com/fossasia/susi_api_wrapper/issues/57). After that, just use `pip` to install `susi_api_wrapper`, don't copy it to `susi_linux`.

woshikie, 2018-07-08:
According to https://github.com/fossasia/susi_linux/blob/master/install.sh
I believe that swig is first installed at
- https://github.com/fossasia/susi_linux/blob/7157726e4ce3afd75a5098122a6dffc1c7086b03/install.sh#L161
which is defined at
- https://github.com/fossasia/susi_linux/blob/7157726e4ce3afd75a5098122a6dffc1c7086b03/install.sh#L6-L12

woshikie, 2018-07-08:
@hongquan As explained above, I believe you may check the first point

hongquan, 2018-07-09:
SWIG always exist in OS repo, and already at version 3.0.12, so please remove the step of installing SWIG from source. That step brings the burden of maintaining the additional code, which has very little helpful.

woshikie, 2018-07-09:
@hongquan After digging around a little, I found out that you can install the latest `snowboy` from `npm`. Perhaps we can use npm to install instead of manually building it

Reference: 
- https://www.npmjs.com/package/snowboy
- https://github.com/Kitt-AI/snowboy

hongquan, 2018-07-10:
>  I found out that you can install the latest snowboy from npm.

@woshikie It is to install NodeJS binding of Snowboy. Our project is Python, not NodeJS.

woshikie, 2018-07-10:
Ahh I see. My bad

PR #248 (2018-07-16): Removed swig from dependencies
Completly Fixes #219 

Removed Swig's source installation and using OS repo's to install it
closed 2018-07-16 by woshikie