fossasia/susi_linux #220: Alternative for Tizonia

woshikie, 2018-07-03:
Tizonia is a huge library and is also buggy. Therefore an alternative would be good.

**Suggested Alternative**

- https://github.com/Sorashi/youtube-play
- https://github.com/mps-youtube/mps-youtube

stealthanthrax, 2018-07-03:
I was thinking instead of using separate programs to play just youtube videos, we could just use `mpv --no-video [youtube-link]` to play the music

hongquan, 2018-07-04:
Yes, let's go with `mpv`.

stealthanthrax, 2018-07-04:
I would like to work on this issue

woshikie, 2018-07-05:
looks like there's some codacy issues with #224 
perhaps we could take another approach.
For an example, https://stackoverflow.com/questions/49354232/stream-youtube-audio-from-python-using-url-without-downloading-python

hongquan, 2018-07-05:
The Codacy issue is not related to choosing what program to play. It is the issue of using which function to call external program.

Note, @stealthanthrax, `mpv` is still an external program. The installation script should tell user to install `mpv` as dependency.

stealthanthrax, 2018-07-05:
@woshikie , we can't follow that approach since vlc requires x11 libraries and we'll not be using graphical interfaces