fossasia/susi_linux #223: Media Discovery should not be run under `root`

hongquan, 2018-07-04:
**Describe the bug**
I [discovered](https://github.com/fossasia/susi_linux/issues/219#issuecomment-402525554) that Media Discovery, the script to create skill to play audio file, when USB drive is inserted, is being running under _root_ user. This is not good. Except when our application is a system program (like `systemd`, `ModemManager` etc), we should not let our application run with "root" permission. This is to prevent our bug/mistake from destroy the system.

The reasons that our script is running under "root" are:
- Our script is triggered by `udev`.
- Our script calls the `mount` command, which requires "root" permission.

Some direction to research:
- `udev` seems not to support switching user. But `udev` can be [combined](https://bbs.archlinux.org/viewtopic.php?id=206555) with `systemd` to achieve this.
- Use [`udisk`](https://wiki.archlinux.org/index.php/Udisks) to mount USB drive as non-root user.

hongquan, 2018-07-05:
Test again, I found that the _auto_skills.py_ script doesn't try to mount USB drive. It calls "mount" without argument, just to list all mount points. This action doesn't require "root".

But there is another issue, this script assumes that the USB drive was automatically mounted by some other program (like Desktop Environment). So this script fails to discover media files if Rasp Pi is run in headless mode (just console, no Desktop Environment).

I think this script must support headless Raspberry Pi, because when we deliver SUSI.AI devices to end-user, we may not include DE. The device is just dedicated to run SUSI.AI.

stealthanthrax, 2018-08-23:
This issue is fixed now, media daemon doesn't require root permissions

closed 2018-08-23 by stealthanthrax