fossasia/susi_linux #229: Create proper vector SVG image

hongquan, 2018-07-05:
**Describe the bug**
@prateekiiest , when @mariobehling tell you to make diagram document in SVG, he wants a vector image, so that it comes as "source", can be edited, and can be scale. But why you make it in bitmap and just embed to SVG file? This is not honest.

**Expected behavior**
The file is a true SVG image, with lines and shapes being made by SVG code.

**Screenshots**
Current SVG source:

[image]

[image]

prateekiiest, 2018-07-10:
@hongquan , @mariobehling did not assign me this issue for creating the svg . This was assigned to @stealranthx during one of the previous meetings as far as I remember. So he has made the update accordingly here #213.

So either it might have got changed while updating the base branch in one of my PRs or it has been updated by some one else. This happened without any intention of mine. Even if it was an accidental commit, I will make a PR soon to update the changes.

hongquan, 2018-07-10:
Sorry, @prateekiiest.

prateekiiest, 2018-07-19:
@hongquan @stealthanthrax please have a look at #290

PR #290 (2018-08-13): Update the SUSI Smart Speaker Workflow
Fixes #229
The workflow svg
closed 2018-08-13 by Orbiter