fossasia/susi_linux #234: Update the installation docs regarding configuring and running SUSI on the Raspberry Pi

Akshat-Jain, 2018-07-06:
I've been trying to setup Raspberry Pi 3 for several days now, and finally managed to configure and run SUSI on it yesterday. I think the installation docs are missing a few important steps. 
Hence, the installation docs should be updated with the proper procedure for configuring and running SUSI on the Raspberry Pi.

woshikie, 2018-07-06:
Can you tell us what are the extra steps you took to install on your raspberry pi 3?

Akshat-Jain, 2018-07-06:
@woshikie All the steps mentioned in the installation doc need to be executed once we're in the `SUSI.AI` folder, right? I can't find this info in the docs, and I actually spent a lot of time trying to execute the commands in the root directory itself. Also, the `install.sh` command should be run with `sudo` otherwise I was getting errors like permission denied. I think these minor things should be added in the docs. Thanks :-)

woshikie, 2018-07-06:
the install.sh should **not** be ran with `sudo` because the `SUSI.AI` is not supposed to be touching any system settings or files

Akshat-Jain, 2018-07-06:
@woshikie Running it without `sudo` was ending the script with `permission denied` error for me.

stealthanthrax, 2018-07-06:
@Akshat-Jain , `sudo` issue is fixed now we can use it as it is

PR #235 (2018-07-07): Updated the installation docs
Fixes #234 
Added a few steps in the installation docs, based on the issues I faced while configuring and running SUSI on Raspberry Pi 3.
closed 2018-07-07 by Orbiter