fossasia/susi_linux #237: Avoid hard-code full path in Python code

hongquan, 2018-07-06:
One example code

```py
def make_skill(): # pylint-enable
    name_of_usb = get_mount_points()
    print(type(name_of_usb))
    print(name_of_usb[0])
    x = name_of_usb[0]
    os.chdir('{}'.format(x[1]))
    USB = name_of_usb[0]
    mp3 = [file for file in glob("*.mp3")]
    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/media_daemon'))
    f = open('custom_skill.txt','w')
    music_path = list()
    for mp in mp3:
        music_path.append("{}".format(USB[1]) + "/{}".format(mp))

    song_list = " ".join(music_path)
    skills = ['play audio','!console:Playing audio from your usb device','{"actions":[','{"type":"audio_play", "identifier_type":"url", "identifier":"file://"'+str(song_list) +'}',']}','eol']
    for skill in skills:
        f.write(skill + '\n')
    f.close()
    shutil.move('/home/pi/SUSI.AI/susi_linux/media_daemon/custom_skill.txt','/home/pi/SUSI.AI/susi_linux/susi_server/susi_server/data/generic_skills/media_discovery')
    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/susi_server/susi_server/data/settings/'))
    f2 = open('customized_config.properties','a')
    f2.write('local.mode = true')
    f2.close()
    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/'))
```

It will take a lot of effort to fix the code if we move susi_linux to other location later. Please:

1. Replace those hard-code path with relative path, or absolute path built with `os.path` functions and special constant `__file__`. For example, the script above is _susi_linux/media_daemon/auto_skills.py_:

```py
# To get media_daemon folder
media_discovery_folder = os.path.dirname(os.path.abspath(__file__))
# To get susi_linux/susi_server/susi_server/data/generic_skills/media_discovery
susi_linux_folder = os.path.dirname(media_discovery_folder)
server_skill_folder = os.path.join(susi_linux_folder, 'susi_server/susi_server/data/generic_skills/media_discovery')
```

2. Stop moving around with `os.chdir`. It is easy to lost track and confused with "current working folder". Use the full path to write/read file. Don't "chdir" then read/write.

You can apply the same strategy with Bash code.

stealthanthrax, 2018-07-06:
@hongquan , there some bash scripts that when run through python , which identify `/home/pi` path as `root` if we do not hardcode `/home/pi`

eg python server files

I'd like to fix the rest issues

hongquan, 2018-07-07:
@stealthanthrax Can you show me which script? Based on specific case, we can have appropriate solution.

woshikie, 2018-07-13:
Here are some files containing "SUSI.AI" word in it, possibly hinting an absolute path

```
fallenwings@ubuntu:~/susi_linux$ ack SUSI.AI
media_daemon/99-susi-usb-drive.rules
3:ACTION=="add", KERNEL=="sd?", SUBSYSTEM=="block", ENV{ID_BUS}=="usb", RUN="/home/pi/SUSI.AI/susi_linux/media_daemon/autostart.sh"
4:ACTION=="remove", KERNEL=="sd?", SUBSYSTEM=="block", ENV{ID_BUS}=="usb", RUN="/home/pi/SUSI.AI/susi_linux/media_daemon/autostop.sh"

media_daemon/autostart.sh
5:cd $HOME/SUSI.AI/susi_linux/
8:    cd $HOME/SUSI.AI/susi_linux/media_daemon/

media_daemon/auto_skills.py
14:    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/media_daemon'))
25:    shutil.move('/home/pi/SUSI.AI/susi_linux/media_daemon/custom_skill.txt','/home/pi/SUSI.AI/susi_linux/susi_server/susi_server/data/generic_skills/media_discovery')
26:    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/susi_server/susi_server/data/settings/'))
30:    os.chdir(os.path.expanduser('~/SUSI.AI/susi_linux/'))

media_daemon/DAEMON.md
10:The `autostart.sh` acts as a starting point for executing the `auto_skills.py` which in turn creates the custom skill under '$HOME/SUSI.AI/susi_linux/susi_server/susi_server/data/generic_skills/media_discovery/' as custom_skill.txt . The custom skill contains a JSON response of the audio play skill with file identifiers USB name and mp3 file name. 

media_daemon/autostop.sh
3:cd $HOME/SUSI.AI/susi_linux/media_daemon

factory_reset/factory_reset.sh
4:cd $HOME/SUSI.AI/susi_linux

access_point/server/server.py
21:    subprocess.call(['sudo', 'bash', '/home/pi/SUSI.AI/susi_linux/access_point/server/login.sh', auth, email, passwd]) #nosec #pylint-disable type: ignore
28:    subprocess.call(['sudo', 'bash', '/home/pi/SUSI.AI/susi_linux/access_point/wifi_search.sh', wifi_ssid, wifi_password])  #nosec #pylint-disable type: ignore

access_point/server/login.sh
6:cd /home/pi/SUSI.AI/susi_linux/

access_point/server/config.sh
7:cd /home/pi/SUSI.AI/susi_linux

access_point/wap.sh
10:APSSID="SUSI.AI"

README.md
8:SUSI AI on Linux
25:- Configuring led lights in accordance with SUSI.AI response

main/ui/glade_files/signin.glade
8:    <property name="title" translatable="yes">Login to SUSI.AI</property>

susi_server/susi_skill_data/models/general/Knowledge/en/messenger_bots.txt
37:!expect:I had an amazing chat with SUSI.AI! Do you want to try it too?
38:I had an amazing chat with SUSI.AI! Do you want to try it too?
42:!expect:You can visit the SUSI.AI repository and start contributing.
43:You can visit the SUSI.AI repository and start contributing.
50:#The below queries are shown as examples of what the user can try asking SUSI.AI

susi_server/susi_skill_data/models/general/Knowledge/en/aboutsusi.txt
4:::description Information about SUSI.AI 

susi_server/susi_skill_data/models/general/Knowledge/en/creator_info.txt
6:::description Want to know about fossasia, just ask susi to tell that, Susi tells about the SUSI.AI creators

susi_server/susi_server/.yaydoc.yml
31:    - url: https://github.com/fossasia/MMM-SUSI-AI

susi_server/susi_server/html/index.html
175:      <h2 style="font-weight: 300; font-size: 25px;">SUSI.AI API Server for Artificial Intelligence Personal Assistants, Robots, Help Desks and Chatbots</h2>
177:          This server provides an output of intelligent answers in JSON format for SUSI.AI clients.

susi_server/susi_server/html/images/susi-logo.svg
17:   sodipodi:docname="SUSI.AI_Letters_Blue_Large.svg">

susi_server/susi_server/src/ai/susi/server/api/aaa/ResendVerificationLinkService.java
98:                EmailHandler.sendEmail(emailId, "SUSI AI verification", getVerificationMailContent(token, identity.getName()));

susi_server/susi_server/src/ai/susi/server/api/aaa/SignUpService.java
216:				EmailHandler.sendEmail(signup, "SUSI AI verification", getVerificationMailContent(token, identity.getName()));

susi_server/susi_server/src/ai/susi/server/api/vis/MarkdownServlet.java
192:        PrintTool.print(matrix, matrix.getWidth() - 6, matrix.getHeight() - 6, 0, "MADE WITH HTTP://SUSI.AI", 1, false, 50);

susi_server/susi_server/conf/templates/resend-verification-link.txt
1:You just requested for a new verification link for your SUSI AI account.

susi_server/susi_server/conf/templates/reset-mail.txt
2:initiate the password reset process for your SUSI.AI account, click the link below:
12:The SUSI.AI Team

susi_server/susi_server/conf/templates/verification-mail.txt
1:You just signed up for SUSI AI!
7:If you didn't sign up for SUSI AI, please ignore this email.

susi_server/susi_server/README.rst
2:SUSI.AI Server
208:If you simply want to add your skill to the SUSI.AI online service, please go to https://skills.susi.ai and add your skill.

susi_server/susi_server/docs/skills/susi_questions.md
1:# SUSI.AI

docs/raspberry-pi_install.md
31:- In your root directory, make a folder named `SUSI.AI` and move into the folder.
34:$ mkdir SUSI.AI
35:$ cd SUSI.AI
91:    - Authenticate to SUSI.AI
```

woshikie, 2018-07-16:
Fixed with #258

closed 2018-07-16 by woshikie