fossasia/susi_linux #243: To correct the workflow of the Installation Script

stealthanthrax, 2018-07-08:
Current Behavior
Right now we do not have a way to execute the installation script . We cannot run the install script while the app is connected to the speaker as there is no Internet access to app. After the reboot (which is essential to allow Internet access) , the app is now connected to the Internet but is not connected to the mobile and hence the installation script cannot be executed.  

Expected Behavior

I have 2 possible approaches

1> To add the install.sh script in the `autoboot` folder when the last script is executed while the phone and device are connected and remove the `install.sh` script from `autoboot` after the installation has been completed.
(Can be a bit complex and error prone)

2> To keep the python server running all the time and then making the mobile device connect again to the smart speaker again after Internet has been established and then run the `install.sh` script.  This will also increase the possibility to control `pause/play/stop` functionality in the future.
(To confirm with the mobile team weather this approach is possible) 

Would you like to work on this issue 
Yes

stealthanthrax, 2018-07-08:
@mariobehling @Orbiter @cweitat @woshikie @hongquan @prateekiiest  , do you have any other suggestions for the approaches?

@batbrain @imjog @arundhati24 , can you check if the second approach is possible?

woshikie, 2018-07-09:
Perhaps we should group the installation process into different groups:

- Basic Installation
    - Cloning Git Repoitory
    - Running installation script

This should always be done because the user needs internet connection to clone the github repository
- Configuration
    - Through CLI
    - Through Android App

Configuration should be able to be done through either Android App or CLI **with or without** internet connection

hongquan, 2018-07-09:
Sorry @stealthanthrax, I don't understand you very well.

> We cannot run the install script while the app is connected to the speaker as there is no Internet access to app

What do you mean by "the app"? Which app? Why "there is no Internet access to app"?

> the app is now connected to the Internet but is not connected to the mobile and hence the installation script cannot be executed.

Why "not connected to the mobile" causes "installation script cannot be executed"?

I think the "install.sh" script is for developer, not end-user. Developer has many ways to access to Pi to run the script, not limit to mobile phone.

What we deliver to end-user is a Raspberry Pi with pre-installed SUSI.AI and almost everything has been working, except that it needs to be reconfigured Wifi in order to connect to user's home network.

What we are stuck now is that, even developer cannot finish installation with _install.sh_ script and make _susi_linux_ working on developer's Pi.

stealthanthrax, 2018-07-09:
@hongquan , I am referring to SUSI mobile app and there is no Internet access to smart speaker as it is in WAP mode.

Okay , I get it now.
>What we deliver to end-user is a Raspberry Pi with pre-installed SUSI.AI and almost everything has been working, except that it needs to be reconfigured Wifi in order to connect to user's home network.

Please tell me what the issues we are facing with the installation on the pi?
>What we are stuck now is that, even developer cannot finish installation with install.sh script and make susi_linux working on developer's Pi.

hongquan, 2018-07-09:
> Please tell me what the issues we are facing with the installation on the pi?

Ok. Just got updated that other developers have successfully installed it.

For myself, I'm running headless Raspbian, so the USB Flash drive is not mounted automatically and "auto_skills.py" still fails.