fossasia/susi_linux #246: Runtime Errors

woshikie, 2018-07-09:
**Describe the bug**
Runtime Errors/Warnings

**To Reproduce**
Steps to reproduce the behavior:
1. Run `python3 -m main` from the repository folder

**Screenshots**
- On Raspberry Pi Zero W
    - [Installation Log](https://pastebin.com/d88gHx4t)
    - [Configuration Log](https://pastebin.com/KKGCSbZz)
    - [Runtime Log](https://pastebin.com/HXWGcWnk1)

woshikie, 2018-07-09:
@stealthanthrax 
Does the logs for Pi Zero looks okay to you? 
Does your Raspberry Pi outputs a similar log?
I do not currently have a microphone or speaker to test for the Pi Zero

woshikie, 2018-07-09:
For installation on Linux Mint 18.3, it looks like install.sh is stuck after running  
https://github.com/fossasia/susi_linux/blob/8dbacb101b5b6c8601f8b0808daff6aa25f0e4d5/install.sh#L145
and not running the codes after it

woshikie, 2018-07-09:
While SUSI is running on Linux Mint 18.3, it looks like SUSI only recognizes "SUSI" and not the words afterwards.
I have tried saying "SUSI tell me a joke" and ""SUSI tell me the time" but it appears to me on the log that only the word SUSI is recognized. 
I have also realized that the GUI is removed

woshikie, 2018-07-09:
While SUSI is running on Ubuntu 18.04 LTS, it appears to me the SUSI is not detecting even `SUSI` 
Could it be because that the local server can't be built and that it has caused some issue?

woshikie, 2018-07-09:
I have successfully built `susi_server` on Ubuntu 18.04 LTS and it appears to me that `susi_server` does not need to be running and `susi_linux` is able to detect `susi` word, but same as on Linux Mint 18.3, `susi_linux` does not seem to detect any other words.

woshikie, 2018-07-09:
I realized that if I run `python3 -m main`, there is no GUI but if i run `python3 app.py`, there's GUI but functionality-wise, it is experiencing the same issue as explained above

woshikie, 2018-07-09:
Regarding the above comment, I realized that the behaviour is expected as documented in the install guide for both [raspberry pi](https://github.com/fossasia/susi_linux/blob/master/docs/raspberry-pi_install.md)
and [ubuntu](https://github.com/fossasia/susi_linux/blob/master/docs/ubuntu_install.md)

stealthanthrax, 2018-07-09:
@woshikie , that is because snowboy isn't really supported on Ubuntu and pocket sphinx sometimes gives unreliable results

>I have successfully built susi_server on Ubuntu 18.04 LTS and it appears to me that susi_server does not need to be running and susi_linux is able to detect susi word, but same as on Linux Mint 18.3, susi_linux does not seem to detect any other words.

stealthanthrax, 2018-07-09:
@woshikie , instead of using `python3 config_generator.py stt tts hotword wake` , you have to choose choices as the default service provider. I can now see the confusion. I'll make a PR about it.

So you'll have to use `python3 config_generator.py google google y n` , where 'y' is the choice if we want use Snowboy as the service provider and 'n' is if we have no wake button connected in the GPIO port

woshikie, 2018-07-09:
> `@woshikie , instead of using python3 config_generator.py stt tts hotword wake , you have to choose choices as the default service provider. I can now see the confusion. I'll make a PR about it.
So you'll have to use python3 config_generator.py google google y n , where 'y' is the choice if we want use Snowboy as the service provider and 'n' is if we have no wake button connected in the GPIO port

Yes it is very confusing. Please use <placeholders> properly (with angle brackets). 

>And regarding the Runtime Logs , I get the same log. And after your Hotword 'SUSI' has been detected , you'll hear the sound and the answers will be processed.
I get the exact same log when its working fine

Great! We're finally getting closer to be on the same page.

closed 2018-07-13 by woshikie