fossasia/susi_linux #257: The prebuilt whl package of snowboy seems to have wrong name

hongquan, 2018-07-13:
**Describe the bug**
I built snowboy on my RPI 3 to wheel package and uploaded to https://repo.fury.io/fossasia/. The file is named _snowboy-1.3.0-py3-none-any.whl_, but I suspect that this name is wrong. This package contains some _*.so_ binary, compiled from C/C++ code. That _*.so_ file will be different between ARM and x86, so the name suffix of package cannot be _none-any_. It should be _linux_armv7l_.

With the suffix _none-any_, I worry that pip on PC will mistakenly download this file and install, while the file was built for ARM7.

**What to do**

Try to fix _setup.py_ build script.

woshikie, 2018-07-13:
On my ubuntu 18.04, the cache looks like 
```
fallenwings@ubuntu:~$ tree ~/.cache/pip/wheels
/home/fallenwings/.cache/pip/wheels
├── 38
│   └── 80
│       └── 4f
│           └── ddc3e8c2b788f2c7f1d625ae870f6bafd3038ff04a3445a2f8
│               └── pocketsphinx-0.1.3-cp36-cp36m-linux_x86_64.whl
├── 47
│   └── 75
│       └── 23
│           └── e52a191eb8609b8cb4b0a369b0246cbe848651f63281f42ba8
│               └── web_cache-1.0.2-cp36-none-any.whl
├── 4b
│   └── 4c
│       └── 89
│           └── 0ac8da441c95da916b458bd62ab79ccd4ecdfa27d01d837bdd
│               └── pyudev-0.21.0-cp36-none-any.whl
├── 78
│   └── 85
│       └── e2
│           └── b34f719fea77c7a1585c91cf8c85c7032e0f9345f82b50e300
│               └── async_promises-1.1.1-cp36-none-any.whl
├── 81
│   └── cd
│       └── de
│           └── 4e0916f623c2d125502e493394fd333ed693960264d4b7e524
│               └── watson_developer_cloud-1.5.0-cp36-none-any.whl
├── 90
│   └── 74
│       └── b8
│           └── 362c8d8e9fefe2fc2a31881c272053eccae6fd22ecd461f672
│               └── pyalsaaudio-0.8.4-cp36-cp36m-linux_x86_64.whl
├── b3
│   └── 76
│       └── f7
│           └── 85353c829c0881e74b5366ce0ed59042b098bb4903e2da8828
│               └── Twisted-18.4.0-cp36-cp36m-linux_x86_64.whl
├── bf
│   └── c9
│       └── a3
│           └── c538d90ef17cf7823fa51fc701a7a7a910a80f6a405bf15b1a
│               └── future-0.16.0-cp36-none-any.whl
├── c0
│   └── a1
│       └── 27
│           └── 5ba234bd77ea5a290cbf6d675259ec52293193467a12ef1f46
│               └── pycparser-2.18-py2.py3-none-any.whl
├── c4
│   └── 94
│       └── 43
│           └── 406fd144e7a4f67099698d9438a10a89351dca69867b2f2fe2
│               └── google_speech-1.0.16-cp36-none-any.whl
├── c6
│   └── b2
│       └── d2
│           └── be6785a207eaa58d76debc10c9d5c66196b40a88abb61d6af7
│               └── zope.interface-4.5.0-cp36-cp36m-linux_x86_64.whl
├── de
│   └── c2
│       └── 93
│           └── 660fd5f7559049268ad2dc6d81c4e39e9e36518766eaf7e342
│               └── python_Levenshtein-0.12.0-cp36-cp36m-linux_x86_64.whl
└── f4
    └── a8
        └── a4
            └── 292214166c2917890f85b2f72a8e5f13e1ffa527c4200dcede
                └── PyAudio-0.2.11-cp36-cp36m-linux_x86_64.whl

52 directories, 13 files
```
it looks like there are some cross-platform libraries and some platform-specific libraries

hongquan, 2018-07-13:
Normally, the pure-Python packages have _none-any_ suffix.

@woshikie, you don't have snowboy on your laptop? What is its name?

stealthanthrax, 2018-07-13:
@hongquan , we do not have any `setup.py` script

woshikie, 2018-07-13:
@hongquan I have made some changes and it's compiling into correct names on my ubuntu 
https://github.com/Kitt-AI/snowboy/pull/475

@stealthanthrax we're trying to build `snowboy` into a wheel package for we can use pip3/pip to install from our own repository

woshikie, 2018-07-13:
@hongquan here's a compiled wheel for armv6, compiled using Pi Zero W
[snowboy-1.3.0-cp35-cp35m-linux_armv6l.whl.zip](https://github.com/fossasia/susi_linux/files/2193946/snowboy-1.3.0-cp35-cp35m-linux_armv6l.whl.zip)

hongquan, 2018-07-14:
@woshikie You have full permission on Gemfury repo. Please upload the file.

By the way, I'm looking for the better way to fix the setup.py script. Learning from other projects who also have Python modules built from Swig, they all use `distutils.core.Extension` or `setuptools.extensions.Extension` to build C/C++ code and hence, generating correct name.

Example:
- http://git.savannah.gnu.org/cgit/libcdio/pycdio.git/tree/

hongquan, 2018-07-16:
> we do not have any setup.py script

I meant snowboy's [setup.py](https://github.com/Kitt-AI/snowboy/blob/master/setup.py) script.

hongquan, 2018-08-30:
Fixed by https://github.com/Kitt-AI/snowboy/pull/493

closed 2018-08-30 by hongquan