fossasia/susi_linux #261: Error in susi_state_machine.py

stealthanthrax, 2018-07-14:
**Describe the bug**
Main script breaking after snowboy refractor

**To Reproduce**
Install as usual
1. Install as usual
2. `python3 -m main`

**Expected behavior**
To bring the script in working state

**Log**
`
File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/SUSI.AI/susi_linux/main/__main__.py", line 4, in <module>
    susiStateMachine = SusiStateMachine()
  File "/home/pi/SUSI.AI/susi_linux/main/states/susi_state_machine.py", line 88, in __init__
    components = Components(renderer)
  File "/home/pi/SUSI.AI/susi_linux/main/states/susi_state_machine.py", line 59, in __init__
    from main.hotword_engine import SnowboyDetector
ImportError: cannot import name 'SnowboyDetector'

`
**Additional context**
Add any other context about the problem here.

stealthanthrax, 2018-07-14:
@prateekiiest , since you have been working Snowboy ,it would be faster if you could have a look here

prateekiiest, 2018-07-14:
Ok I will be looking into it tonight

prateekiiest, 2018-07-14:
Solving this will solve the issue here
https://github.com/fossasia/susi_linux/blob/67aaa44cb1a885a0254e64d7f46d3fa3044b9d42/main/hotword_engine/snowboy_detector.py#L11

stealthanthrax, 2018-07-15:
Fixed at #262 #263

closed 2018-07-15 by stealthanthrax