fossasia/susi_linux #296: Media Discovery Daemon Not working

stealthanthrax, 2018-07-22:
**Describe the bug**
USB skills not being created as the USB is inserted

**To Reproduce**
Run the install script and inserted USB

**Expected behavior**
Automatic USB skills are created

**Additional context**
I would like to work on it

stealthanthrax, 2018-07-25:
@hongquan @woshikie @cweitat , can you check if its the same with you guys. Because when I run `udevadm test` if get this error `syspath parameter missing`

hongquan, 2018-07-26:
Where is the log?

stealthanthrax, 2018-07-26:
@hongquan , I am unable to get the error log since when I run `udevadm test`, I get the error `syspath parameter missing` .But when I insert USB, custom skills are not made

hongquan, 2018-07-27:
Get the log where you see `syspath parameter missing`.

How can people know what happen with `syspath parameter missing` if you don't provide all information around it?

hongquan, 2018-07-27:
About "custom skills are not made", did you read this: https://github.com/fossasia/susi_linux/issues/223#issuecomment-402600426?