fossasia/susi_linux #307: Support more audio file type in Media Discovery

hongquan, 2018-07-27:
**Is your feature request related to a problem? Please describe.**
Currently, Media Discovery only looks for mp3 files in USB drive. The media player can play more than that, especially open source format (Ogg Vorbis, FLAC, Opus etc.).

**Describe the solution you'd like**
- Just change the `glob('*.mp3')` code to list more file.
- If possible, retrieve the list of supported file types from `sox` itself.

prateekiiest, 2018-08-01:
some thing as such structure might me good

```
music_path = list()

mp3files = glob('*.mp3')
oggfiles = glob('*.ogg')
flacfiles = glob('*.flac')

for mp in mp3files:
        music_path.append("{}".format(USB[1]) + "/{}".format(mp))

for mp in oggfiles:
        music_path.append("{}".format(USB[1]) + "/{}".format(ogg))

for mp in flacfiles:
        music_path.append("{}".format(USB[1]) + "/{}".format(flac))

    song_list = " ".join(music_path)  # song list will contain all such songs
```

stealthanthrax, 2018-08-13:
This feature has been implemented [here](https://github.com/fossasia/susi_linux/pull/316)

closed 2018-08-13 by stealthanthrax