fossasia/susi_linux #309: Implement action types pause, resume and restart

Orbiter, 2018-07-29:
**Describe the solution you'd like**
If a media file is played, it is possible to say "stop" to stop playing. With the following action types, it should be possible to
* **pause** the playing of a media file (youtube, mp3 etc) with the action type `pause`
* **resume** the playing of a media file which has been paused with the action type `resume`
* **restart** the playing of a media file (either currently playing or paused) with the action type `restart`

**Additional context**
Examples:
`https://api.susi.ai/susi/chat.json?q=pause`
```
"actions": [{"type": "pause"}]
```

`https://api.susi.ai/susi/chat.json?q=resume`
```
"actions": [{"type": "resume"}]
```

`https://api.susi.ai/susi/chat.json?q=restart`
```
"actions": [{"type": "restart"}]
```

stealthanthrax, 2018-08-01:
I would like to work on it

stealthanthrax, 2018-08-01:
@Orbiter , right now we pause the Music to process the second query. Now we would have to follow a different approach. Should we be reducing the volume of the music player instead??

hongquan, 2018-08-03:
> have to follow a different approach

@stealthanthrax Can I learn more about the context? Which approach is that? Where is it described?

tmccormi, 2018-10-16:
Adding an action to change the volume up and down seems to be relevant for this feature request as well

PR #427 (2018-12-13): Added pause/resume/restart actions
Fixes #309 

Dependent on PR : https://github.com/fossasia/susi_api_wrapper/pull/74
closed 2018-12-13 by hongquan