fossasia/susi_linux #319: Make the Python client a proper PyPI package

felixonmars, 2018-08-05:
**Is your feature request related to a problem? Please describe.**
Currently the susi client tries to install into a specific home location (/home/pi/...) which is prohibited by any system-level packaging system.

**Describe the solution you'd like**
By using a PyPI package system (setup.py and setuptools, entry points) this can be standardized and make distribution packaging much easier.

System-level dependencies can be resolved as it is now in the install.sh, and with distribution packaging they will be much easier for the users.

prateekiiest, 2018-08-05:
Related to #236

felixonmars, 2018-08-05:
It's related, and IMHO the work on a PyPI package will greatly make the others simpler.

prateekiiest, 2018-08-05:
Agreed :+1:

hongquan, 2018-08-06:
FYI, the location `/home/pi/SUSI.AI` is just temporarily for now.
We will decide the location later, but need to research at once with packaging method, reset mechanism. All these features depend on, and affect the choice of installation location.