fossasia/susi_linux #323: Reduce OS image size

hongquan, 2018-08-07:
This issue is to track what we need to do to reduce image size, so that it is suitable for release.

The strategy is to:

- Use prebuilt binary of 3rd-party software as much as possible (avoid downloading its source and building on RPi).
- For softwares that the original author doesn't provide a binary, we build them ourselves and host in our repository.
- If the official repo (APT, PyPI) has needed 3rd-party software, but in older version than what we need, we build the newer version ourselves and put to our repo.
- (To be added later)

Checklist:

- [x] Setup PyPI repository to host Python wheel packages.
- [ ] Setup APT repository to host softwares which don't exist in Debian official repo.
- [ ] Use Debian package of _flite_.
- [ ] Build Java SUSI server to binary and host on our repo. I'm not sure if this is possible. Need comment from @Orbiter.
- (To be added later)

hongquan, 2018-08-07:
This article worth reading when we are creating image file: https://source.tizen.org/documentation/reference/bmaptool/introduction

hongquan, 2018-08-07:
Raspbian image build script seems to be this one: https://github.com/RPi-Distro/pi-gen

(there are some scripts on GitHub but I'm not sure which one is the official yet).

mariobehling, 2018-08-19:
As temporary work-around try to use e.g. 7zip and part 1, part 2 etc. functionality.

hongquan, 2018-08-20:
We include *susi_server*, which needs to download _gradle-4.2.1_ to build itself. The _gradle_ contributes around 160MB.

hongquan, 2018-08-20:
Java alone contributes ~ 250MB

```
$ sudo apt install ca-certificates-java
Reading package lists... Done
Building dependency tree        
Reading state information... Done
The following additional packages will be installed:
  java-common libavahi-client3 libcups2 liblcms2-2 libnspr4 libnss3 libxi6
  libxrender1 libxtst6 openjdk-9-jre-headless x11-common
Suggested packages:
  default-jre cups-common liblcms2-utils fonts-dejavu-extra
  fonts-ipafont-gothic fonts-ipafont-mincho ttf-wqy-microhei | ttf-wqy-zenhei
  fonts-indic
The following NEW packages will be installed:
  ca-certificates-java java-common libavahi-client3 libcups2 liblcms2-2
  libnspr4 libnss3 libxi6 libxrender1 libxtst6 openjdk-9-jre-headless
  x11-common
0 upgraded, 12 newly installed, 0 to remove and 24 not upgraded.
Need to get 108 MB of archives.
After this operation, 247 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

hongquan, 2018-08-20:
Git database of *susi_server* is around 200MB.

```
susi_server$ du -hd 1
232K	./docs
468K	./conf
24K	./build
68K	./gradle
3.0M	./html
156K	./test
64K	./kubernetes
205M	./.git
12K	./ssi
2.1M	./src
8.0K	./.bluemix
60K	./bin
24K	./.settings
8.0K	./dependencies
212K	./.gradle
8.0K	./.utility
16K	./.github
8.0K	./scripts
524K	./installation
212M	.
```

hongquan, 2019-01-07:
Done.

closed 2019-01-07 by hongquan