fossasia/susi_linux #330: To install 3rd-party software from binary form

hongquan, 2018-08-17:
This is part of attempt to reduce OS image (#323).

In our _install.sh_ script we are trying to install some dependencies by building from source, whereas they can be installed from prebuilt binaries. We will try to replace the install method as much as possible.

This issue is to track which softwares I can change the installation method.

- [x] flite

hongquan, 2018-08-17:
Trying to find binary release of _seeed-voicecard_. Found that respeaker has a Debian repo here https://github.com/respeaker/deb, but not sure if it contains _seeed-voicecard_

hongquan, 2018-08-17:
Looking in _seeed-voicecard_'s install script, I feel that it is not easy to convert it to a Debian package, so just leave it installed from source.

hongquan, 2018-08-17:
seeed-voicecard's install script pulls down `libasound2-plugins`, which, in turn, pulls a lot of video-related stuff: mesa-va-drivers, mesa-vdpau-drivers, x11-common, even though voicecard just deals with audio.

prateekiiest, 2018-08-20:
so is there a way we can bypass the installation of `libasound2-plugins` ?
I see the only option to send a PR to the main https://github.com/respeaker/seeed-voicecard repo to add the functionality of choice if we need to install `libasound2-plugins`  or not

hongquan, 2019-01-07:
Done.

closed 2019-01-07 by hongquan