fossasia/susi_linux #331: To Open the StartUp play sound with the python module

stealthanthrax, 2018-08-18:
**Is your feature request related to a problem? Please describe.**
Right Now the startup sound is played as an external service but I would like it to be played every time SUSI Linux is started. This will also reduce the hindrance provided due to the service bootup.

**Describe the solution you'd like**
To play it from the python program, every time we start SUSI Linux.

**Additional context**
I would like to take this up.

stealthanthrax, 2018-08-18:
@mariobehling @Orbiter @hongquan , should we implement it this way ??

hongquan, 2018-08-18:
I thought "every time SUSI Linux is started" is the same "every time the RPi is boot up"?

For the final product, does user have any method to restart the "susi linux" alone, without reboot the RPi hardware?

stealthanthrax, 2018-08-19:
No @hongquan , the user does not have to start SUSI Linux independently , but I was using it for the development

PR #341 (2018-08-22): Added startup audio
Fixes #331 

Plays StartUp sound on SUSI starting
closed 2018-08-22 by hongquan