fossasia/susi_linux #355: Config flask server is not working after removal of wrappers

stealthanthrax, 2018-08-27:
**Is your feature request related to a problem? Please describe.**
After the shell wrappers were removed, the flask server doesn't create config.json file.

**Describe the solution you'd like**
To fix it.

**Additional context**
Yes

mariobehling, 2018-09-01:
Use journalctl to see the route from systemd and provide more info. Currently it is unclear how to reproduces this.

mariobehling, 2018-09-01:
It stopped working after PR https://github.com/fossasia/susi_linux/pull/334 was merged.

mariobehling, 2018-09-01:
Related to https://github.com/fossasia/susi_linux/issues/324

prateekiiest, 2018-09-02:
I have tried reverting the changes but its not possible I see. Somehow it has been reverted I see from the github message

prateekiiest, 2018-09-02:
@stealthanthrax did you locate why the error is happening?

We need to check any sort of dependancy between the flask server creation of json file wrt to the wrapper script removal

stealthanthrax, 2018-09-02:
@prateekiiest , I still am trying to figure it out. It most probably looks like a directory issue.

PR #369 (2018-09-09): Fixing the local server and config.json bug
Fixes #355 

Fixes the scripts after wrappers were removed.
closed 2018-09-09 by hongquan