fossasia/susi_linux #363: susi_linux failed to start on boot

hongquan, 2018-09-01:
**Describe the bug**
susi_linux failed to start when RPi boot up.
From the log (https://paste.ee/p/CyiJ4), there are some issues:

- Try to connect to SUSI server when SUSI server is not ready (not listen port 4000 yet).
- Try to play audio when Jack server is not ready yet.
- Crash when microphone is not connected.

hongquan, 2018-09-01:
It seems that ReSpeaker driver is not installed properly on my RPi.

hongquan, 2018-09-02:
@stealthanthrax Why not look into the log I pasted above:

> Sep 01 19:20:59 raspberrypi python3[540]: Cannot connect to server socket err = No such file or directory
> Sep 01 19:20:59 raspberrypi python3[540]: Cannot connect to server request channel
> Sep 01 19:20:59 raspberrypi python3[540]: jack server is not running or cannot be started
> Sep 01 19:20:59 raspberrypi python3[540]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

stealthanthrax, 2018-09-02:
@hongquan , I think we can ignore those logs as they are an issue from Snowboy side and not on our side and SUSI work's fine with these logs

hongquan, 2018-09-06:
@stealthanthrax You should specify which issue can be ignored.
With the 3rd issue, SUSI Linux crashes, we cannot ignore, because how come it work when it failed to start?

stealthanthrax, 2018-09-06:
@hongquan , i was referring to this issue
>Try to play audio when Jack server is not ready yet.

stealthanthrax, 2018-09-15:
@hongquan , can this be closed now??

hongquan, 2018-09-16:
Close, because the problems no longer happen, and the "jack server" seems to be PyAudio's false report.

closed 2018-09-16 by hongquan