fossasia/susi_linux #390: web search responses are not working

kasojumounika, 2018-09-30:
susi ai speaker could not get information from web search.

**To Reproduce**
Steps to reproduce the behavior:
1.say susi.
2. What is famous in Hyderabad.
3. It just processes the web query but doesn't give the answer.

**Expected behavior**
As i questioned what famous in hyderabad.i expected susi to answer Charminar. Charminar is a historic monument in Hyderabad. ...
Falaknuma Palace. Falaknuma Palace. ...
Qutub Shahi Tombs.

**Screenshots**
NA

**Desktop (please complete the following information):**
NA

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kasojumounika, 2018-09-30:
I would like to work on this issue.

stealthanthrax, 2018-10-03:
sure ,@kasojmounika :+1:

stealthanthrax, 2019-02-22:
I would like to work on this due to inactivity.

stealthanthrax, 2019-02-23:
This issue has been fixed by a server update and hence closing it.

closed 2019-02-23 by stealthanthrax