fossasia/susi_linux #393: To add a way to create automated Raspbian images via QEMU and Travis

stealthanthrax, 2018-10-03:
**Is your feature request related to a problem? Please describe.**
To fix the problem of manually creating images with win32imager

**Describe the solution you'd like**
An approach suggested by @xeon-zolt during jugaadfest to use qemu with travis and release a new image on every PR.

**Additional context**
[image]

hongquan, 2018-10-03:
The purpose of this issue is duplicate with https://github.com/fossasia/susi_linux/issues/382

stealthanthrax, 2018-10-31:
Update:

The QEMU scripts seem to work fine now. There is some issue with the install scripts and would be fixed within a week now

hongquan, 2018-11-01:
You keeps saying that your QEMU script works fine, but until now, I still don't see any evidence to show that "it works".

The Travis log in #392 show that it even *cannot* start the virtual machine, let alone run the install script.

stealthanthrax, 2018-11-01:
@hongquan , you can check here https://api.travis-ci.org/v3/job/449021181/log.txt , you will see that after qemu has started the raspbian OS, it crashes out in an "alsa" problem coming via the installation script.

hongquan, 2018-11-01:
@stealthanthrax That log shows that the `qemu-system-arm -kernel ./kernel-qemu-4.4.34-jessie -append...` command failed, not the install script.

Because, the beginning of the install script is the "apt-get update" command. If the `qemu-system-arm` passed, and the install was run, there must be output of `apt-get update` in the log.

hongquan, 2018-11-01:
To compare, this is the build log of my PR #410:

- The line 15615 is where the _./install.sh_ script started to run.
- The line 24704 is where _cmu_us_slt.flitevox_ was downloaded.
- The line 24824 is where _susi_server_ was cloned.
- The line 27627 is where the image file was generated and being compressed.
- The line 26134 is where Gradle finished to build susi_server
- The line 26282 is the message which marked that installation is finished ("SUSI is installed successfully").

[build-susibian_build_1_config-spiral-note-pad-and-run-building-construction-build-dot-sh.log.gz](https://github.com/fossasia/susi_linux/files/2538691/build-susibian_build_1_config-spiral-note-pad-and-run-building-construction-build-dot-sh.log.gz)

Any where in your logs show those steps?