fossasia/susi_linux #399: Add support for pocketsphinx for wakeword detection

ftyers, 2018-10-14:
**Is your feature request related to a problem? Please describe.**

It's frustrating to rely on non-free software.

**Describe the solution you'd like**

Let's use free software for the wakeword detection, like pocketsphinx or something else.

prateekiiest, 2018-10-14:
we are using pocketsphinx for stopword detection

ftyers, 2018-10-14:
I understood that [snowboy](https://snowboy.kitt.ai/) was being used. Perhaps I am mistaken?

prateekiiest, 2018-10-14:
As for the time being, we are using both options depending on whether the user has snowboy installed or not.

https://github.com/fossasia/susi_linux/blob/87d4495bf0041d6cab85af06c5de24153e3b1cb3/main/hotword_engine/__init__.py

so you can switch to any one of them, upon your wish.

The pocketsphinx implementation is here https://github.com/fossasia/susi_linux/blob/62b23b1147a8ef22bb08f2ad97b0772e3681d43e/main/hotword_engine/sphinx_detector.py

PR #411 (2018-11-04): Fixed the sensitivity issue in SUSI hotword
Fixes #399 

Fixes fuzzy hotword detection for the default piHat extension boards and removes the excess files
closed 2018-11-04 by hongquan