fossasia/susi_linux #402: Wakeword detection on RPI Device is too fuzzy

tmccormi, 2018-10-16:
Any number of sounds that are not the word SUSI pronounced by me include parts of english sentences and even random humming (which I do a lot) wake up the device.

To reproduce, just have one on in the room and have normal conversations with people 

**Expected behavior**
Wakeup should only happen (mostly) with a clearly recognized word.

**Desktop (please complete the following information):**
None

**Smartphone (please complete the following information):**
N/A

Suggestion:  Make the wake word two words (like google does)  "OK SUSI" or whatever.

tmccormi, 2018-10-16:
Seriously, clearing my throat triggers SUSI :-)

stealthanthrax, 2018-10-17:
@tmccormi , a possible fix would be to reduce the sensitivity of the hotword detection to a lower scale and train the model with different accents

stealthanthrax, 2018-11-04:
Fixed here : [link](https://github.com/fossasia/susi_linux/issues/399)

closed 2018-11-04 by stealthanthrax